SiEPIC/SiEPIC_EBeam_PDK #15: Specify WG length

lukasc-ubc, 2015-11-21:
Once drawn, give user ability to type in a length, and have waveguide adjust itself.

Add Length as parameter in waveguide PCell.  Gettable & Settable.

jaspreetj, 2016-04-10:
Works but needs improvement. There are some bugs eg. sometimes the layout disappears upon the completion of the resizing operation.

press Ctrl + Shift + R to run the script while having a transient selection.

lukasc-ubc, 2016-04-10:
Hi Jaspreet,

A few problems:

 - Layout disappears.  I've seen this before when using a GUI and performing operations.  In my case, the solution was to close the GUI first, then perform layout modifications

 - if you select the waveguide, it crashes.  only works if you select the path.  should work for both.  see my script "Waveguide_to_path"

 - remove 1st and last segments; these presumably never will be useful.

 - perhaps only list eligible segments: name those whose neighbour segments point in the same direction (middle of a U-shape being the only allowable one).

lukasc-ubc, 2016-04-11:
fixed:

 - layout disappears

 - selection for PCell vs. GUIDING_PATH; both work

 - correctly measuring the path length including bends

 - added Undo functionality

 - removed 1st/last points.

to do:

 - quality control checking

 - perhaps only list eligible segments: name those whose neighbour segments point in the same direction (middle of a U-shape being the only allowable one).

lukasc-ubc, 2016-04-11:
Bug:

 - if the segment to be moved is on the left, it applies the "diff" in the incorrect direction (makes the waveguide shorter instead of longer)

jaspreetj, 2016-04-11:
working on the Bug and to do items.

jaspreetj, 2016-04-11:
**Fixed: if the segment to be moved is on the left, it applies the "diff" in the incorrect direction (makes the waveguide shorter instead of longer)**

New Bug detected: Waveguide resizer does not work on a waveguide which has already been resized.

lukasc-ubc, 2016-04-11:
Bug: segment says "horizontal" but it is vertical.

lukasc-ubc, 2016-04-12:
did some GUI improvements.

jaspreetj, 2016-04-13:
The script was giving me syntax errors. So I did following fixes. 

I commented out the line no. 110.

and changed line no. 79 from
`if len(o_selection)<>1:`
 to
`if len(o_selection)< 1:`

PR #72 (2016-04-10): update
- MC for Original GCs
- Resize Waveguide Fix #15
commit d6433fbb (2016-04-10): Merge pull request #72 from jaspreetj/master

MC for Original GCs; Resize Waveguide Fix #15
closed 2016-04-21 by jaspreetj