SiEPIC/SiEPIC_EBeam_PDK #33: CML - TM polarization for components

lukasc-ubc, 2015-12-05:
- TM GC does not excite the TM mode.

 - Y branch missing the TM simulations.

strip waveguide model correctly has TE and TM modes.

121comeon, 2016-04-08:
For the TM GC, we are using the Optical S-Parameter element. Currently, its property "Waveguide > modes" is "TM", but it should be " , TM", so that ONA's orthogonal identifier 1 & 2 corresponds to "TE" & "TM". 

In general, it is better to use the Optical N Port S-Parameter element, which is clearer and can include multiple modes (though syntax is a bit longer).

lukasc-ubc, 2016-04-20:
Fixed the GC CML to have Waveguide > modes = ", TM"

Also updating KLayout PDK:
 - added orthogonal_identifier to Laser
 - netlist script to use orthogonal_identifier
 - created a sample layout for TE and TM.

closed 2016-04-20 by lukasc-ubc