SiEPIC/SiEPIC_EBeam_PDK #56: Spice export for INTERCONNECT, netlist trimming

lukasc-ubc, 2016-02-20:
- check that there is only 1 laser.
- search the Netlist and remove components that are not connected to a laser/detector.  Performance improvement especially for MC simulation.

lukasc-ubc, 2016-04-21:
Graph connected components:
 - https://breakingcode.wordpress.com/2013/04/08/finding-connected-components-in-a-graph/

jaspreetj, 2016-04-23:
Do we just want a group of elements or they need to be in a particular order?

for eg. 
Group #1: a, b, c, d, e, f

Can it be like:
Group #1: c, b, a, d, f, e

lukasc-ubc, 2016-04-23:
Order doesn't matter. They just need to end up in the (short) spice export.
The main purpose is to remove the components that aren't part of the
simulation in order to speed up the time consuming import in interconnect.

jaspreetj, 2016-05-04:
working on it. Almost finished it.

lukasc-ubc, 2016-05-06:
Hi Jaspreet,

Thanks for your pull request!  There was a merge conflict, so I just downloaded your file and merged it myself.  Yes, please do add this to the MC Simulations.

Do we need the disconnected ports on the top of the device?  (I don't think so...)

jaspreetj, 2016-05-07:
I just implemented this to the MC simulations and cleared the unused ports. 

Do we still need the **generate_short_Spice_file(topcell, optical_waveguides, optical_components)** function in the SiEPIC Functions?

If not, can I remove the function and add the newly created function to the Functions file?

lukasc-ubc, 2016-06-11:
I think it is working well.

I have merged it into the main function, and removed the incomplete function you mentioned.

thank you!
Lukas

closed 2016-06-11 by lukasc-ubc