SiEPIC/SiEPIC_EBeam_PDK #117: Bug in PDK

jaspreetj, 2016-05-05:
PDK and KLayout is installed properly but some of the scripts are not working properly.

When I tried to convert path to waveguide, I get the following error:
[image]

Even when I used the previous code, it failed to recognize the selection.
 
The following line returns false for all drawn paths: 
`o.shape.is_path()`
where o refers to object_selection.

If I copy paste a waveguide from a layout, the script works but does not work on newly created paths.
Some of the students in the edX course also encountered this error.

jaspreetj, 2016-05-06:
problem still persists.
[image]

jaspreetj, 2016-05-06:
That fixed the issue with the script.
Also, it is must to keep the highlighted option to **Add**. Otherwise the created paths will not be considered as paths and the scripts will generate errors.
[image]

lukasc-ubc, 2016-05-06:
strange, I don’t see this problem.

closed 2016-05-07 by jaspreetj