SiEPIC/SiEPIC_EBeam_PDK #128: EBeam PDK single bus ring model has a problem

GuanhuiWang, 2016-10-17:
Drug and drop the "ebeam_ring_singlebus_te1550" element from the Design Kits folder to the schematic editor runs into a endless loop of asking for s-parameters and adding other copies of this same element in INTERCONNECT. It might be a invalid path problem.

luzeqin, 2016-10-20:
Hi Guanhui,

Yes it is...I am not sure who made this model, which seems to be incomplete. The real model for ring is ebeam_dc_halfring_straight_te1550. Maybe it is time to clean up the CML package.

GuanhuiWang, 2016-10-20:
Thanks Chris. Using the "ebeam_dc_halfring_straight_te1550" elements will end up with double bus rings rather than single bus. But I think we can work with it for most of the cases.

luzeqin, 2016-10-25:
Hi Guanhui,

Yes, I agree that it's better to have a separate model for sing bus ring. I
will work on this when I get a chance this week.

lukasc-ubc, 2016-11-03:
Thanks Chris.