SiEPIC/SiEPIC_EBeam_PDK #137: Redhat linux server: Showing permission error when running interconnect from klayout

rounaksingh, 2017-03-03:
Hi,
I am taking PIC course at RIT. We are using linux servers for lumerical-interconnect and klayout here. Server uses a common tmp directory(/tmp/). While running interconnect simulation from klayout, I found that in pymacro INTERCONNECT.lym creates various files (spi,icp,lsf) in tmp using the topcell_name(actually active cell name). 
`topcell = pya.Application.instance().main_window().current_view().active_cellview().cell`
When the active cell name of two users is same, linux shows permission denied error.

In my point of view it's better to create folders first according to user name & timestamp, or add username in the file_names. In this way it will be helpful for people sharing drives. This will also be useful when deleting tmp files.

Thanks

lukasc-ubc, 2017-03-04:
good idea.  If you want to experiment with making changes, there is a Python script file "INTERCONNECT".  on line 84, there is code that specifies the path.  you could add the time stamp there.

lukasc-ubc, 2017-03-04:
excellent! Can you make a Pull Request so I can add this in my copy?

closed 2017-03-04 by lukasc-ubc