SiEPIC/SiEPIC_EBeam_PDK #141: Automatic layout save option before running interconnect simulation

rounaksingh, 2017-03-06:
Hi,
This feature is available in most of the modern simulation tool. Please let me know what you think about it

lukasc-ubc, 2017-03-06:
Good idea.
However, this is more of a KLayout issue.  Could you please post this on:
http://klayout.de/forum/ <http://klayout.de/forum/>

Lukas Chrostowski, P.Eng., Professor
Program Director, NSERC Silicon Electronic-Photonic Integrated Circuits (SiEPIC) Program <http://www.siepic.ubc.ca/>
"Silicon Photonics Design" textbook is now available <http://www.cambridge.org/us/academic/subjects/engineering/electronic-optoelectronic-devices-and-nanotechnology/silicon-photonics-design-devices-systems>.
course on edX <http://bit.ly/UBCxPhot> – Silicon Photonics Design, Fabrication and Data Analysis

University of British Columbia
Electrical and Computer Engineering <http://www.ece.ubc.ca/>,
4039-2332 Main Mall, Kaiser Building
Vancouver, B.C., Canada, V6T-1Z4
Office: Room 4039 Kaiser Building
Work: 604-822-8507
http://www.mina.ubc.ca/lukasc <http://www.mina.ubc.ca/lukasc>

 <http://bit.ly/SiPbook> <http://bit.ly/SiPbook> <http://bit.ly/UBCxPhot>

lukasc-ubc, 2017-03-07:
Auto-save before simulation -- that actually makes sense and we could add this to the INTERCONNECT python script...  if we can figure out how to do this...

lukasc-ubc, 2017-03-07:
https://www.klayout.de/forum/comments.php?DiscussionID=933

rounaksingh, 2017-03-07:
I have actually figured it out. But let's wait for the klayout expert to tell a new way.

1) We can get the filename using filename() function of current CellView. 
http://www.klayout.de/0.20/rba_CellView.html#filename
Also, we can create a 
2) Then we can write the file using write function
https://www.klayout.de/doc/code/class_Layout.html#m_write

lukasc-ubc, 2017-03-08:
Looks like it is even more simple: 

RBA::Application::instance.main_window.cm_save

closed 2017-03-08 by lukasc-ubc