SiEPIC/SiEPIC_EBeam_PDK #156: Segfault on KLayout 0.25 on Linux

nsmith5, 2017-07-14:
Not sure if this is a KLayout or Linux thing but `SiEPIC` is giving a segfault and klayout refuses to start on the latest development version. I'll post a screenshot shortly.

nsmith5, 2017-07-14:
[image]

nsmith5, 2017-07-14:
To reproduce: 
 - Build Klayout from source
 - Download SiEPIC compressed archive
 - Copy SiEPIC dot files into `~/.klayout`
 - Run `klayout`

lukasc-ubc, 2017-07-14:
I've never built klayout myself. Can you try the compiled version on their page?
Are you using the latest siepic library? Or a release version?

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

nsmith5, 2017-07-14:
Same on both versions unfortunately. No compiled version for Fedora 26 and the Centos version requires older versions of ruby than my system is capable of supplying.

lukasc-ubc, 2017-07-24:
Hi again,

Does it segfault when you start klayout by itself (without SiEPIC)?  Can you narrow it down to a particular script that causes it to crash?

bpoulse, 2017-07-30:
This looks like it could possibly be a problem with the QT Binding module built for Ruby.

When you built klayout, did you use this option?
build.sh -with-qtbinding ...

Reference:
http://www.klayout.de/doc/programming/qt_binding.html

closed 2018-01-08 by lukasc-ubc