SiEPIC/SiEPIC_EBeam_PDK #158: Problem using shift+O in klayout

cmekhiel, 2017-07-27:
I get an error when I try to snap to objects together in klayout. 
[image]

[EBeam_cmekhiel2.zip](https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/files/1178343/EBeam_cmekhiel2.zip)

Thank you

jaspreetj, 2017-07-27:
Seems like you are trying to snap two components after flattening the layout. You can only snap objects when they are not flattened. This is why you are getting the error.

cmekhiel, 2017-07-27:
Hi, I have tried it without flattening and I still get the error.

lukasc-ubc, 2017-07-27:
can you please upload the non-flattened layout so we can debug?

lukasc-ubc, 2017-07-27:
i just improved the snapping code to provide more error messages when the object isn't an instance. namely can't align flattened polygons/shapes.

cmekhiel, 2017-07-28:
I also have this issue with a new layout trying to snap 2 waveguides where nothing is flattened.

cmekhiel, 2017-07-28:
It looks like I have this issue when I try to snap a path instance to an object from the library. I am able to snap 2 objects together-a directional coupler and a y branch . But I can't use shift+O to snap a path to the Y-branch. Should I always be using the path to waveguide function instead?

lukasc-ubc, 2017-07-28:
we don't have a function to snap two waveguides to each other, unfortunately.  I can think of a way to do that; it would probably take 2 hours of coding to add a snap and merge waveguide feature.

path to object snapping: also doesn't work. use path to waveguide which does what you need.

closed 2017-07-28 by lukasc-ubc

lukasc-ubc, 2017-08-02:
some feedback on your layout:

https://courses.edx.org/courses/course-v1:UBCx+Phot1x+2T2017/discussion/forum/b77035e9087eb9f3b8c43d8ec5abba3f59067576/threads/5980dd82a358c60a5d000235 <https://courses.edx.org/courses/course-v1:UBCx+Phot1x+2T2017/discussion/forum/b77035e9087eb9f3b8c43d8ec5abba3f59067576/threads/5980dd82a358c60a5d000235>