SiEPIC/SiEPIC_EBeam_PDK #159: Generalizing the Path to Waveguide Script

bpoulse, 2017-07-29:
I am interested in generalizing the Path to Waveguide script to allow some other common waveguides (ridge and slot) along with a sort of 'builder' where you can specify different waveguide components, their widths, layer to be tied to, etc. such that you can specify any type of more complex waveguide (inverted, etc) along with adiabatic bends, all in one window.

Some preliminary work has been done to the script (see attached images) that visualizes what I mean.

Comments or suggestions are welcome.
[image]
[image]

lukasc-ubc, 2017-07-31:
Can you add this to a fork so we can see?

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

bpoulse, 2017-07-31:
Yes, I will do so shortly. The actual waveguide generation is not quite stable. So I don't push broken code I will fix it before doing so.

bpoulse, 2017-08-01:
@lukasc-ubc  I have pushed the initial rewrite to my fork. I hopefully moved the necessary files. The path to waveguide and waveguide to path scripts/hotkeys should be working. The code is a bit messy. I would like to clean it up and if you are interested possibly do a little reorganization. Have a utils module etc so we can use the regular Python system (ex. from 'SiEPIC.EBeam import PCells' or 'from SiEPIC import utils'). I found it slightly difficult to navigate the functions macro and think it would be much easier to use given some organizing into py modules.

bpoulse, 2017-08-01:
Adiabatic functionality is not yet implemented but slab, ridge, and slot waveguides along with the custom waveguide option should be working.

lukasc-ubc, 2017-08-01:
Just tried copying your fork and running it.  Nice work so far!

1) Created a path, then pushed W.  Received an error:

'NoneType' object has no attribute 'findChild'
  /Users/lukasc2/.klayout/pymacros/SiEPIC/SiEPIC_EBeam_Path_to_Waveguide.lym:73
  /Users/lukasc2/.klayout/pymacros/SiEPIC/Path_to_Waveguide_Menu_Binding.lym:4

it is the hard-coded backslash:
https://stackoverflow.com/questions/16010992/how-to-use-directory-separator-in-both-linux-and-windows

2) the first type of waveguide would be called a "Strip" rather than "Slab"

3) how/where does the user configure the layers to use?   e.g, Ridge

We'll need a technology file somewhere that is technology specific.  i.e., if GSiP is selected, use XX layers.  Any thoughts on how we can implement such global variables?

I started a thread on KLayout forum: https://www.klayout.de/forum/comments.php?DiscussionID=964&page=1#Item_0

4) default param for slot waveguide doesn't work.  change slot width

5) in the GUI, return/enter doesn't work, nor does the Escape button.

6) seems like the waveguide is no longer a PCell, hence it doesn't have a guiding shape which is useful for moving / stretching the waveguide using the Partial edit tool.  Perhaps that's not a big problem since the Shift-W is working, hence we can resize the path instead.  

7) Custom doesn't seem to work?

8) DevRec - one of the reasons I haven't been using the Waveguide_Route function is that it creates a bunch of components (straight sections, bent sections), rather than one waveguide. This has pros and cons.  One of the big cons was that INTERCONNECT simulations take a lot longer when you have tons of individual components, rather than one long waveguide.  We developed the monte carlo method to work on the simple path approach where we export a list of vertices for the long waveguide, rather than exporting all the individual building blocks.

9)

bpoulse, 2017-08-01:
1. The first error should be fixed. I realized that about 5-10 minutes after the first push.

2. A quick fix I can make

3. At the moment, the settings (layers) for the strip, ridge and slot are hard coded.
To this end, I was thinking of adding a few things
3.a. Add layer selection for these three options
3.b. Add another section in the GUI that lets you load a compact model for straight and bent sections
3.c. Allow the user to name and save a configuration. The class will serialize the config and the user can open it from a popup list view the next time.

4. All the defaults should be verified I think. I was not sure what the most common widths were for ridge and slot wg's.

5. The button bindings have yet to be tied to hotkeys

6. Even though the waveguide is no longer a pcell, the path is still stored as a cell property. It might be possible to utilize this. The reason I did not make it a pcell was I could not seem to input an array as a parameter. This made it difficult for specifying multiple layers.

7. I'll check it out

8. I haven't yet looked into how DevRec is used, or how this all ties to the Interconnect binding. To start, I simply tried to emulate how the waveguides were made previously. As mentioned, the cell still has the original path stored as a user property (for simplicity in the waveguide to path script). If this is all that's needed, it can be modified to use this.

bpoulse, 2017-08-03:
@lukasc-ubc can you point me in the right direction on some literature for implementing the bezier curve? I modified the GUI to input the "Bezier parameter" you talked about in the EDX course, but I am not sure how that ties into the parametric equation for a Bezier curve.

lukasc-ubc, 2017-08-03:
We have it in the Siepic Pyxis PDK. Do you have that one?  It is written in a C like language. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

bpoulse, 2017-08-04:
@lukasc-ubc 
Hey Lukas, thank you for the information. I found it. I didn't end up using part of the code that calculated the normals, I reimplemented it by computing the Bezier contour and scaling it around the origin of the bend. I dont think there is a difference, but it was much easier.

Take a look if you have time, everything except connections to Interconnect and such as been implemented (my TODO's are in my readme). Please let me know if you notice any issues or have any comments.

For the technology part. I have figured out how to do this, but it the path-wg script and pcells will need to be reorganized. Is that ok?

closed 2017-09-05 by bpoulse