SiEPIC/SiEPIC_EBeam_PDK #170: cleanup for v0.3.0 EBeam and GSiP PDKs

lukasc-ubc, 2017-12-22:
Hi Mustafa, Jaspreet,

Can you please help as follows?

for GSiP, EBeam, and EBeam-dev, update:

 - refer to ebeam_dc_halfring_arc in EBeam-dev as example
 - arc waveguides for bends: use arc_wg or arc_wg_xy, from SiEPIC.utils
  - replace all layout_arc_wg_dbu with the above
 - from SiEPIC._globals import PIN_LENGTH, for the length of the pin, instead of hard coded
 - ensure all pins are pointing out of the device (order of vertices)

check all components, including waveguide snapping.

mustafacc, 2017-12-23:
Can you take a look this [IME PDK PCell](https://github.com/lukasc-ubc/SiEPIC_IME/commit/df8d2564ee582d81e305049b6d9ebde425dc72c6)? If it's good then I'll proceed with modifying the EBeam pcells on that framework (with ebeam tech.)

Thanks.

lukasc-ubc, 2017-12-25:
Awesome, thanks.  Did some fixes.

closed 2018-02-08 by lukasc-ubc