SiEPIC/SiEPIC_EBeam_PDK #171: Silicon Photonic Device Design Competition

lukasc-ubc, 2017-12-29:
We are hosting a design competition, with a cash prize for the winner. The objective is to design a specific component, given a set of specifications, with a specific optimization goal.

The component must be designed and modelled between February 6, 2018 and March 5, 2018, and submitted for fabrication by March 5, 2018, via the edX Silicon Photonics Design, Fabrication and Data Analysis course.  We will fabricate your devices via Applied Nanotools measure them, and provide you experimental data in mid-March. You will analyze the experimental data, and submit your design into the competition by the final deadline – 2 weeks after all the experimental data is provided (estimated deadline at the end of March or early April).

Design objective: 
    **Broadband, Dual-Polarization, 50/50 splitter.**

Specifications:

- Port 1, input: 500 nm strip waveguide
- Port 2 and 3, two outputs: 500 nm strip waveguides separated by 1 µm
- Broadband operation between 1200-1700 nm
- Dual-polarization: quasi-TE and quasi-TM
- Device can operate as a splitter or as a combiner
- The design must be compatible with modern DUV 193 nm immersion lithography CMOS foundries; namely we are setting the minimum feature size to be 50 nm.  See the following reference which illustrates imec’s fabrication silicon photonics capabilities with features down to 50 nm [1].  [1] S. K. Selvaraja, G. Winroth, S. Locorotondo, G. Murdoch, A. Milenin, C. Delvaux, P. Ong, S. Pathak, W. Xie, G. Sterckx, and G. Lepage, “193nm immersion lithography for high performance silicon photonic circuits," in Proc. SPIE, Mar. 2015, pp. 90520F-1.

Optimization goal:

- Minimize the excess insertion loss, IL = - 10 log10(0.5) + 10 log10 (Port 3 / Port 1)
- Minimize the imbalance in the splitting ratio, IR = abs(10 log10 (Port 3 / Port 2)), for injection in Port 1.
- Figure of Merit (FOM) = 3 x max(ILExperimental, ILSimulated, worst corner) + max(IRExperimental, 

IRSimulated, worst corner)

Submission Requirements:

- A GDS layout file, named SiEPIC-EBeam-competition-<your edX username>.gds
- Simulation files
- Corner analysis to consider fabrication variations with process corners defined as ∆w = +/- 10 nm, and ∆h = +/- 10 nm.
- A compact model, which calculates the device response for any input to any output combination (e.g., input on Port 2, output on Port 1), for one or both polarizations (e.g,. Injection of both TE and TM light in Port 1)
- A compact model that is suitable for Monte Carlo simulations, namely the compact model should be parameterized (e.g., interpolation) within the process corners
- A report containing the above information. You may pursue a publication of work, in which case a draft version of your manuscript can be submitted as your report.
- The component is to be added to the SiEPIC-EBeam-PDK on GitHub as a pull request. 
- You retain ownership of the component, but you grant us license to distribute your component on the SiEPIC-EBeam-PDK repository. 
- The submission needs to include: 

  1. documentation in the folder Documentation/2018T1_Design_Competition,
2) GDS file in the folder klayout_dot_config/tech/EBeam/pymacros, as an add-on to the Library “SiEPIC-EBeam-competition”, 
3) a compact model, SiEPIC-EBeam-competition-<your edX username>.ice, in the folder Lumerical_EBeam_CML/EBeam, 
4) design simulation files in a folder Lumerical_EBeam_CML/EBeam/fdtd/SiEPIC-EBeam-competition-<your edX username>, 
5) A report, the layout containing the test structures that were used to measure the performance, and experimental data, in a folder Examples/SiEPIC-EBeam-competition-<your edX username>, 
6) State the value obtained for the FOM in the pull request comment.

khavasi, 2018-01-31:
Hi
The waveguide width in the provided layout templates is 500 nm but the width of waveguide in our design should be 450 nm according to the competition rules. How can we change the width of waveguides in the template? If it is not easy can we use a tapered waveguide to match the widths?

lukasc-ubc, 2018-01-31:
@khavasi, 
Good point. Will change the template files to use 450 nm waveguides rather than 500 nm. For your device, you can assume the input/output pins will be 450 nm.

lukasc-ubc, 2018-02-08:
@khavasi 
I fixed the pins to be 450 nm in the GDS files.
a369b1f405d2d64765e72e405bed57dda47dbe23

gbfarias, 2018-02-09:
Dear Lukas
From the NanoSOI fabrication process website, it seems like the Oxide deposition (cladding) is optionnal in the process. Could you confim that there will be the oxide deposition in the process?
Thanks!
Giovanni

GerardoCastanon, 2018-02-09:
Dear Lukas 
I was reading the specifications of the Y-junction design competition 
and i have a question about the separation of port 2 and 3.
I wonder if the value of 2 micrometers is correct?, it is a must?
In the reference paper they use 0.2 micrometers. 

Cheers,
Gerardo

lukasc-ubc, 2018-02-09:
@gbfarias Giovanni,

The process we will be using includes oxide.

We will fabricate your design using two facilities: 1) University of Washington, and 2) Applied Nanotools.

Here is a description of each:

Applied Nanotools, Inc. NanoSOI process:

The photonic devices were fabricated using the NanoSOI MPW fabrication process by Applied Nanotools Inc. (http://www.appliednt.com/nanosoi; Edmonton, Canada) which is based on direct-write 100 keV electron beam lithography technology. Silicon-on-insulator wafers of 200 mm diameter, 220 nm device thickness and 2 µm buffer oxide thickness are used as the base material for the fabrication. The wafer was pre-diced into square substrates with dimensions of 25x25 mm, and lines were scribed into the substrate backsides to facilitate easy separation into smaller chips once fabrication was complete. After an initial wafer clean using piranha solution (3:1 H2SO4:H2O2) for 15 minutes and water/IPA rinse, hydrogen silsesquioxane (HSQ) resist was spin-coated onto the substrate and heated to evaporate the solvent. The photonic devices were patterned using a Raith EBPG 5000+ electron beam instrument using a raster step size of 5 nm. The exposure dosage of the design was corrected for proximity effects that result from the backscatter of electrons from exposure of nearby features. Shape writing order was optimized for efficient patterning and minimal beam drift. After the e-beam exposure and subsequent development with a tetramethylammonium sulfate (TMAH) solution, the devices were inspected optically for residues and/or defects. The chips were then mounted on a 4” handle wafer and underwent an anisotropic ICP-RIE etch process using chlorine after qualification of the etch rate. The resist was removed from the surface of the devices using a 10:1 buffer oxide wet etch, and the devices were inspected using a scanning electron microscope (SEM) to verify patterning and etch quality. A 2.2 µm oxide cladding was deposited using a plasma-enhanced chemical vapour deposition (PECVD) process based on tetraethyl orthosilicate (TEOS) at 300ºC. Reflectrometry measurements were performed throughout the process to verify the device layer, buffer oxide and cladding thicknesses before delivery.

Washington Nanofabrication Facility (WNF) silicon photonics process:

The devices were fabricated using 100 keV Electron Beam Lithography [1]. The fabrication used silicon-on-insulator wafer with 220 nm thick silicon on 3 μm thick silicon dioxide. The substrates were 25 mm squares diced from 150 mm wafers. After a solvent rinse and hot-plate dehydration bake, hydrogen silsesquioxane resist (HSQ, Dow-Corning XP-1541-006) was spin-coated at 4000 rpm, then hotplate baked at 80 °C for 4 minutes. Electron beam lithography was performed using a JEOL JBX-6300FS system operated at 100 keV energy, 8 nA beam current, and 500 µm exposure field size. The machine grid used for shape placement was 1 nm, while the beam stepping grid, the spacing between dwell points during the shape writing, was 6 nm. An exposure dose of 2800 µC/cm2 was used. The resist was developed by immersion in 25% tetramethylammonium hydroxide for 4 minutes, followed by a flowing deionized water rinse for 60 s, an isopropanol rinse for 10 s, and then blown dry with nitrogen. The silicon was removed from unexposed areas using inductively coupled plasma etching in an Oxford Plasmalab System 100, with a chlorine gas flow of 20 sccm, pressure of 12 mT, ICP power of 800 W, bias power of 40 W, and a platen temperature of 20 °C, resulting in a bias voltage of 185 V. During etching, chips were mounted on a 100 mm silicon carrier wafer using perfluoropolyether vacuum oil. Cladding oxide was deposited using plasma enhanced chemical vapor deposition (PECVD) in an Oxford Plasmalab System 100 with a silane (SiH4) flow of 13.0 sccm, nitrous oxide (N2O) flow of 1000.0 sccm, high-purity nitrogen (N2) flow of 500.0 sccm, pressure at 1400mT, high-frequency RF power of 120W, and a platen temperature of 350C. During deposition, chips rest directly on a silicon carrier wafer and are buffered by silicon pieces on all sides to aid uniformity.

[1] R. J. Bojko, J. Li, L. He, T. Baehr-Jones, M. Hochberg, and Y. Aida, "Electron beam lithography writing strategies for low loss, high confinement silicon optical waveguides," J. Vacuum Sci. Technol. B 29, 06F309 (2011)

lukasc-ubc, 2018-02-09:
@GerardoCastanon, Dear Gerardo,

The requirement for having the output waveguides be separated by 2 µm is so that the splitter can be used in practical applications, namely where we have two independent waveguides that can be connected to other things.  With the output waveguides spaced at 0.2 µm, the waveguides are actually strongly coupled (like a directional coupler), so one needs to separate them first before connecting to other waveguides. Part of the design challenge is engineering how you split the waveguides apart (and making it compact). Typically people use an S-Bend for that.

You can take a look at the SEM image of the Y-branch above for an example. Also the layout for that ybranch is included in this EBeam PDK.

Thanks for participating!

regards,
Lukas

vatalogg, 2018-02-13:
Dear Professor Lukas,

First, thank you for actively supporting us for the competition. I have got 2 questions, the first for the design and the second for the evaluation.

(A) It is mentioned that "the minimum feature size is 60 nm". How does that affect our design in terms of specific design rules? For example, does it imply something about the minimum distance between two silicon waveguides? Also, if, for any reason, I have a silicon trapezoid should the smallest (parallel) side have a length of at least 60 nm?

Lastly, are these design rules verified through the KLayout (similarly I have seen a DRC check in a VLSI design software) or it is just that our circuits will be distorted if we violate them?

(B) The insertion loss is negative through: IL = - 10 log10(0.5) + 10 log10 (Port 3 / Port 1), with the best value being zero (when Port 3/Port 1=0.5). So, I assume that we should minimize the value of abs(IL). Please correct me, if I misunderstood something.

Sincerely,
Vasileios

khavasi, 2018-02-14:
Hi Lukas,
I encountered a problem in uploading my files. I used this link: http://upload.siepic.ubc.ca/openEBL.php and I received a message that the file has been successfully uploaded. But when I checked these links: http://upload.siepic.ubc.ca/openEBL/openEBL.txt
http://upload.siepic.ubc.ca/openEBL/openEBL_coords.txt?lipi=urn%3Ali%3Apage%3Ad_flagship3_pulse_read%3BrjYJ8yqcSWWuEgfKBHbGSw%3D%3D
I could not find details of my gds file. I tried several names for my gds file (and its top cell) and the last one was openEBL_khavasi_MZI.gds with top cell name openEBL_khavasi_MZI. 
Thank you in advance for your consideration.

lukasc-ubc, 2018-02-14:
@vatalogg, Vasileios,

A) Min feature size of 60 nm, means that the minimum distance between any two pieces of silicon has to be 60 nm or more.  A trapezoid (e.g., nanotaper) can have a tip that is at least 60 nm. Take a look at the file "ebeam_competition2018T1_example.gds" which shows the min feature for the tip being 60 nm.

The rules are there to help ensure that the facility can fabricate what you design. Foundries typically enforce these rules.  Indeed there is a DRC check like in VLSI design.  We have a DRC check file, which is the menu SiEPIC | Verification | DRC. It is presently in the master copy on GitHub, but not yet released through Package Manager.

B) Insertion Loss: Indeed, better minimize abs(IL) the way I (incorrectly) defined it.

lukasc-ubc, 2018-02-14:
@khavasi 

I just turned on the openEBL submission system.  I turn it off on each fabrication run.

I see your openEBL*.gds files are there.

khavasi, 2018-02-14:
Would you mind deleting my files? I'll upload them again later.

lukasc-ubc, 2018-02-16:
Here is a video excerpt from an edX Phot1x Live Q&A, where I discussed how to simulate a component from a layout in KLayout, using Lumerical FDTD. The simulation generates the S-Parameters, which are loaded into INTERCONNECT, and which allows for Circuit Simulations.

https://youtu.be/qWtbPcVbAbY

LuhuaXu, 2018-02-24:
Hi Lukas, I have two questions regarding the three layout templates you provided:
(1) the center to center separation of two output waveguides is 2.5 um, and the edge to edge separation is 2.05um; none of them is 2.0 um according to the design specifications; I am wondering if you could, for example, change the center to center separation of two output waveguides to be 2.0 um? 
(2) there is an 0.1um I/O routing waveguide intrusion into the device region, which makes the actual allowable device length to be 49.8um instead of 50um. I am wondering if you could remove those intrusion or move them by 0.1um so that the the actual allowable device length becomes 50um? I am interested in pursing a publication and an integer-number device length would be preferred

LuhuaXu, 2018-02-24:
Hi Lukas, 
Regarding my first question above, I did a literature review and found a 2um edge-to-edge (i.e. 2.45um center-to-center) separation would be better, due to the weak confinement of TM modes

lukasc-ubc, 2018-02-24:
Dear Luhua

Based on what you read about waveguide separation, what I have drawn (2.5 center to center) should be sufficient. 

In terms of the space, you don’t need a 100 nm waveguide intrusion. What is needed are pins on layer 1/10, but these are not fabricated (they are only used for identification). So you have the full length available. 

Thanks for participating!

LuhuaXu, 2018-02-24:
Hi Lukas, thanks for your reply! Please correct me if what I understand is wrong:
The pins with length of 0.2 um on layer 1/10 are only for identification purpose, and they will not be fabricated; only structures on layer 1/0 will be fabricated.

lukasc-ubc, 2018-02-25:
@LuhuaXu : Correct.

LuhuaXu, 2018-02-27:
Hi Lukas, I am wondering if there is any restriction if I want to  purse a journal publication for the device fabricated in this competition?

Thanks!

LuhuaXu, 2018-02-27:
@lukasc-ubc 
Hi Lukas, I am also wondering if the SEM images could be provided for the fabricated device?

Thanks!

vinhinpl-gist, 2018-03-02:
Dear Professor Lukas,
Is it required that my design should be novel idea? In other words, can I utilize some idea from a published paper?
Thank you,
Vinh Nguyen

lukasc-ubc, 2018-03-02:
@LuhuaXu,

Journal publication: of course!

SEM: yes, we can take a picture of your design.  To request an SEM image, please place a box over the region of interest on layer 26.

lukasc-ubc, 2018-03-02:
@vinhinpl-gist,

Nothing novel is required. You can use published designs and ideas if you wish.

vinhinpl-gist, 2018-03-02:
@lukasc-ubc,

Thank you for your information.

lukasc-ubc, 2018-03-02:
All,

I was pointed out to me that the template files I provided had an inconsistent filename compared to the instructions above.  I have updated this, as openEBL_competition2018T1_TE_<your_GitHub_username>.gds

In commit 281df17def1d1b380f3118ff7e253c935353b392

LuhuaXu, 2018-03-02:
@lukasc-ubc 
Thanks! I am wondering if I can take multiple SEM images (around 4)?

lukasc-ubc, 2018-03-02:
@LuhuaXu,

Each image costs us about $25. So unless there is a good reason to take more pictures, I'd like to provide 1 image. If you would like to pay for more, we can arrange that.

vinhinpl-gist, 2018-03-02:
@lukasc-ubc 
Can I submit 2 different designs if they are both satisfied the competition requirements?

lukasc-ubc, 2018-03-02:
@vinhinpl-gist,

Multiple different designs: yes.

vinhinpl-gist, 2018-03-02:
@lukasc-ubc 

Yes. Thank you a lot.

LuhuaXu, 2018-03-02:
@lukasc-ubc 
Thanks a lot!

lukasc-ubc, 2018-03-04:
@vinhinpl-gist, @LuhuaXu, @khavasi, @vatalogg, @gbfarias, and others,

A reminder about the deadline for submitting to the competition: March 5, 2018, 9:00 pm Pacific Time

Please check the openEBL.gds file to make sure your design files are in there: http://upload.siepic.ubc.ca/openEBL/openEBL.gds

I look forward to seeing your submission!

regards
Lukas

tahmidhassan, 2018-03-05:
Having a really hard time renaming the ports and pins inside the template

lukasc-ubc, 2018-03-05:
Leave the port names as-is

YRumaldo, 2018-03-06:
Dear Lukas,
Please, ignore only version A of OpenEBL_competitionYRumaldo, because it is with connection error
Thank you very much
Yesica

vinhinpl-gist, 2018-03-06:
@lukasc-ubc 
Thank you for your reminder. However, for your information we decide not to submit the design since it almost reuses the idea of published paper. Now I am just a beginner of the integrated photonics field. I have learned a lot through this competition. I hope I can expand my knowledge on the field and submit my design for future competition if any.
Best Regards,
Vinh Nguyen

lukasc-ubc, 2018-03-06:
Yesica / @YRumaldo,
You want me to delete *_YRumaldo_A*?  But leave _B and _C?

YRumaldo, 2018-03-06:
@lukasc-ubc 
Yes, I want to leave B and C.
Best Regards

lukasc-ubc, 2018-03-21:
Dear competitors,

Hossam has measured all the TE devices from the EBeam fabrication.  Files available on

Dropbox link: https://www.dropbox.com/sh/whp1bsljqpsofzw/AABXOj_PZPj-__SYhfMQIiuEa?dl=0

You may download the entire data set by replacing the =0 with =1.  
 
TM devices should be measured later this week.

khavasi, 2018-03-21:
Hi Lukas,
Thank you very much for this interesting competition. I sent four designs whose file/cell names were ....._khavasi1_..., ....._khavasi2_..., ....._khavasi3_... and ....._khavasi4_... . But, unfortunately the labels of grating couplers were the same. Now, the result files have similar names like these:
competition_2018T1_khavasi_N112_TE_1559
competition_2018T1_khavasi_N112_TE_1599
competition_2018T1_khavasi_N112_TE_1792
competition_2018T1_khavasi_N112_TE_1795
Can I assume that they are corresponding to khavasi1,2,3, and 4, respectively?
Best regards,
Amin

YRumaldo, 2018-03-21:
Hi Lukas,

I sent two designs whose names were 'YRumaldo_B' and 'YRumaldo_C'  but I'm
not finding the experimental results.  Have not you measured it yet?

Best regards,

Yesica

-- 
Yesica Raquel Rumaldo Bustamante
Master, Eng.
...........................................................

University of Campinas (UNICAMP)
School of Electrical and Computer Engineering (FEEC)
Dept. of Microwaves and Optics (DMO)
Cidade Universitária Zeferino Vaz
Barao Geraldo CEP 13083-970
Campinas Sao Paulo Brazil
Mobile:+55-19-988922387

lukasc-ubc, 2018-03-21:
A video showing how to fetch experimental results and run a simulation:

https://youtu.be/oRt5bTERGOc

lukasc-ubc, 2018-03-21:
@khavasi 
You can load the matlab files and look in the metadata where the motor coordinates are found.  Hopefully these will give you a hint as to which file is which.

In general, we do require that labels are unique, so that you find your measurements more easily.

lukasc-ubc, 2018-03-21:
@YRumaldo 
Looks like you didn't replace "githubusername" with your own name.  Check your opt_in labels.  Then either use the new feature in SiEPIC-Tools to fetch the measurements, or go to: https://github.com/lukasc-ubc/edX-Phot1x
and search for githubusername...

YRumaldo, 2018-03-21:
Dear Prof. Lukas
Indeed I have not replaced the opt_in labels with my githubusername, so how
can I identify my measurements?
I have seen your video for fetching results in Klayout, but how can I load
the SiEPIC tools in Klayout?
Many thanks for your support.
Yesica

-- 
Yesica Raquel Rumaldo Bustamante
Master, Eng.
...........................................................

University of Campinas (UNICAMP)
School of Electrical and Computer Engineering (FEEC)
Dept. of Microwaves and Optics (DMO)
Cidade Universitária Zeferino Vaz
Barao Geraldo CEP 13083-970
Campinas Sao Paulo Brazil
Mobile:+55-19-988922387

lukasc-ubc, 2018-03-21:
If you are familiar with KLayout, easier for you to fetch the data manually from Dropbox. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

YRumaldo, 2018-03-21:
> Mobile:+55-19-988922387 <(19)%2098892-2387>

Dear prof. Lukas,

We are trying to find our measurements by looking into the <gds_x_coord>
and <gds_y_coord> coordinates specified in the metadata of the .mat files
you supplied, and comparing with the latest GDS on the link you provided
(openEBL.gds), on the main cell.
However, something is not correct because in most cases, the x,y
coordinates specified in the .mat files do not correspond to the correct
cell in the GDS (other users). Is it possible that we are not looking to
the latest GDS file that was used as a reference for the tests?

Thank you for your support
Yesica

-- 
Yesica Raquel Rumaldo Bustamante
Master, Eng.
...........................................................

University of Campinas (UNICAMP)
School of Electrical and Computer Engineering (FEEC)
Dept. of Microwaves and Optics (DMO)
Cidade Universitária Zeferino Vaz
Barao Geraldo CEP 13083-970
Campinas Sao Paulo Brazil
Mobile:+55-19-988922387

lukasc-ubc, 2018-03-21:
Indeed there is an offset where the openebl file was merged into a larger one. The rotation is 0, so it should be possible to infer the offset. Namely look at the relative offsets in your devices. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

Sandra0000, 2018-03-22:
@ YRumaldo 
I am also one of those who forgot to change  the githubusername in opt_in labels. You might be able to  find your designs by checking the coordinates of the designs which are next to yours in the GDS file.  Good luck to you!
Sandra

lukasc-ubc, 2018-03-26:
Dear competitors, @Sandra0000, @YRumaldo, @khavasi, @vinhinpl-gist, @tahmidhassan, @vatalogg, @LuhuaXu, @gbfarias, @GerardoCastanon,  

Hossam has measured all the TE and TM devices from the EBeam fabrication. Files available on

> Dropbox link: https://www.dropbox.com/sh/whp1bsljqpsofzw/AABXOj_PZPj-__SYhfMQIiuEa?dl=0

The deadline for the design competition submission #2 –  April 10, 2018, 9:00 pm Pacific Time.  Report/slides, as described above.

khavasi, 2018-03-27:
Dear Prof. Lukas,
I have a question about this sentence:
"The wavelength range for the analysis is defined as a wavelength span of 50 nm centred at the grating couplers’ peak response." 
The wavelength of grating coupler's peak response is nominally 1550 nm, but in experimental results different wavelengths are obtained for grating coupler's peak response (e.g. 1557 nm in one of my results). Which one do we consider in our analysis, the nominal value or the experimental one?

khavasi, 2018-03-27:
Another question:
How can we convert extinction ratio of MZI to splitting ratio? When I use my own analysis I obtain strange results. For example, for the extinction ratio of 20 dB I obtain splitting ratio of 57:43. Inasmuch as my device (and almost all of other's) is completely symmetric I expect much balanced splitting ratio.

lukasc-ubc, 2018-03-27:
@khavasi,

Splitting ratio: This analysis provides a worst-case.  The ER in the measurements won’t be infinity, so the best I can think of is that extracted splitting ratio will be at worst 57:43.

GC peak response: use the experimental peak to define the range of wavelengths for the analysis, rather than the as-designed. Since the peak wavelength will have some variability, choose an average value.

khavasi, 2018-03-28:
Dear Lukas,
Thank you for your response.  I think directly including splitting ratio (SR) in FOM is not reasonable. Because the splitting ratio will be around 55:45 (in my estimation this is the best result), which is equivalent to SR = 0.87 dB. On the other hand, the insertion loss (IL) is around 0.05 dB for best devices. This means that the importance of IL in FOM is negligible and the FOM is mainly determined by SR. Moreover, in my opinion, the experimental SR for symmetric splitters (which is the case for almost all of competitor's devices) is mainly determined by the measurement limitations rather than device properties (I concluded by investigating many of devices in this competition). 
The second problem is that accurate calculating SR from ER needs the experimental S-parameters of splitters which is not available. We can make some estimations to calculate  SR, but different competitors may use different formulas (I use equation (6) in Khavasi, Amin, et al.  IEEE Photonics Technology Letters 28.24 (2016): 2787-2790.). I think, a unique formula should be specified for all competitors.

lukasc-ubc, 2018-03-28:
@khavasi: good points. Indeed for a symmetric device there is little reason to expect a large SR penalty.  So the best case would be a SR = 0 dB, though we do not have experimental evidence of this.  You can include this discussion in your submission report.

LuhuaXu, 2018-04-02:
@lukasc-ubc Hi Lukas, I am wondering what type of simulation files do you require to submit for submission 2?

lukasc-ubc, 2018-04-04:
@LuhuaXu,

We don't need the simulation files.  Just the plots themselves.  Imagine you were writing a conference submission and had to make some slides and write a short paper.

I did say it is optional that you can submit the simulation files, particularly if you can provide an S-Parameter compact model.

devika001, 2018-04-11:
@lukasc-ubc, Dear Lukas, Due to internet connectivity issue I was unable to upload report and presentation by time. Deadline time is passed and I suppose I am unable to participate now.
Am I allowed to use the results at least for a publication?
Thank you.

khavasi, 2018-04-29:
Hi Lukas,
Could you estimate when the winner will be announced?
Thank you!

lukasc-ubc, 2018-04-29:
@devika001, yes you can use them. 

@hkavasi, thanks for asking. Hopefully this week. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

[1 more reply]

closed 2023-11-21 by lukasc-ubc