SiEPIC/SiEPIC_EBeam_PDK #174: KLayout integration doesn't fully work for KLayout version 0.25

GuanhuiWang, 2018-01-08:
The integration lacks some buttons for KLayout version 0.25 such as the "configure shortcut key" button.
[image]

lukasc-ubc, 2018-01-09:
Thanks Guanhui,

It also looks like you are running an older version of the PDK.  

Can you please upgrade?  Delete the old one, and install the new one using Tools | Package Manager, by selecting EBeam PDK.

The initialization is automatically run on start-up. Can you please check if the hotkeys work?  

e.g., w, v, i, ...

thank you

jflueckiger, 2018-01-09:
Lukas, 
the shortcuts work for me with the latest (pulled 5mins ago). 
J

GuanhuiWang, 2018-01-11:
Thanks Lukas. I am using CentOS 7. Might did something wrong, but when I removed all the old files and recopied the new ones, all the integration buttons were gone. It is very possible that I messed it up, will do some more tests tomorrow.

What I did to get the above picture in the first place was just updated KLayout. The config shortcut key button was there for KLayout v 0.24. 
[image]

lukasc-ubc, 2018-01-11:
Hi again,

Could you please try installing it using the Package Manager, rather than copying files manually?  Also please check File | Log view and let me know what you see there (screenshot can be pasted here).

Instructions:
https://github.com/lukasc-ubc/SiEPIC-Tools/wiki/Installation

mustafacc, 2018-01-11:
I'm seeing the following errors when installing the package on my Linux (Ubuntu 16.04.3)
[image]

And this is the result on my Windows 10 machine
[image]

JeffSteidle, 2018-01-11:
Hi Mustafa,

I ran into the same errors on my Windows 10 machine earlier in the week. After digging through the files I found that the cause was the version of Lumerical Interconnect that was installed. The errors went away after I updated Interconnect.

-Jeff

mustafacc, 2018-01-11:
Thanks, Jeff, I see that updating Interconnect to 7.5 has resolved my issue.

lukasc-ubc, 2018-01-12:
Wow, thank you both for figuring this out.

lukasc-ubc, 2018-01-12:
Hi Mustafa,

To address the Linux system problem:

I released SiEPIC-Tools v0.3.8.  Please update to that one.  Then remove the Windows_Python_Packages, which seems to be causing an issue where it conflicts with the Linux install of numpy/multiarray.

Let me know if that works!
Lukas

lukasc-ubc, 2018-01-13:
> The package manager no longer depends on the Windows Python package, but I'm seeing the following error now:

You still need numpy. It can be installed either via the Windows package, or some other means. 
I removed the dependency since there is no mechanism to only make it for Windows. OS X and Linux doesn’t need it and it was causing problems for Linux. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

GuanhuiWang, 2018-01-16:
Hi Lukas,

Sorry for the late reply, finally got around to this. I tested it on a clean VM (CentOS 7), the installation basically went through but with some "Stream has unknown format" error. Then after installation, the integration buttons are there still without the "config shortcut key" button. Attached files are installation error, Layout after installation and log viewer, respectively. 
[image]
[image]
[image]

lukasc-ubc, 2018-01-17:
All that looks ok and as expected. 

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

MohamedElRayany, 2018-02-07:
I followed the instructions and still I have some issues. I tried to explain them in this [thread](https://github.com/lukasc-ubc/SiEPIC-Tools/issues/62).

closed 2018-02-08 by lukasc-ubc