SiEPIC/SiEPIC_EBeam_PDK #176: Changing pinRec layer

mustafacc, 2018-01-24:
Should we change/standardize pinRec layer on EBeam to match GSiP on SiEPIC-Tools? It would be nice to click and drop EBeam components on GSiP technology layout and have the whole layout be compatible and ready to go to INTC.

lukasc-ubc, 2018-01-24:
Makes sense.  Also to be consistent with SiEPIC_IME_Library.  
Waveguide: 1/0
Waveguide Pin: 1/10

The nice thing is that doesn’t affect any of the fabrication, since these are just for internal use.

mustafacc, 2018-01-26:
https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/commit/5b2b2d4ab604c79b994873793feb6563d47e3c5e

lukasc-ubc, 2018-01-26:
We need to update all the GDS components in the library, the PCells, as well as all the examples. .. !

closed 2018-02-07 by lukasc-ubc