SiEPIC/SiEPIC_EBeam_PDK #180: Pin Issue: ebeam_gc_te1550

StephenHLin, 2018-02-06:
The pins on the gc seem to have issues with the new path_to_waveguide function.
I've tried this both manually through the GUI and through scripts.

I suspect perhaps the pins are drawn in the wrong direction.

mustafacc, 2018-02-06:
What's the issue exactly? Is is the pin_length mismatch? They are snapping fine on my end.

StephenHLin, 2018-02-06:
[image]

This is the result on my end. Were the gc's pin lengths updated just now? I'm seeing 0.2 right now

mustafacc, 2018-02-06:
It doesnt affect snapping as far as I'm aware, but it is fixed now:
https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/commit/f584faf4d91046fe4d768157e4428c894b64c2b3

closed 2018-02-06 by mustafacc