SiEPIC/SiEPIC_EBeam_PDK #184: Export netlist without GCs

smartalecH, 2018-03-10:
Can we re-enable the feature to export the netlist without requiring a grating coupler?

Our research group is exploring other coupling capabilities, and we would like to look at a circuit's response without having to normalize out the GCs' responses.

jaspreetj, 2018-03-13:
Yes it is possible. 

One way to do it would be to provide an option when launching circuit simulations to have GCs in the layout but exclude them from the exported netlist.

your thoughts @lukasc-ubc ?

lukasc-ubc, 2018-03-19:
Just added this.

To use it, use the old method of putting down a Laser and Detectors in your circuit.  In the Laser PCell, choose “Ignore optical IOs in simulations” = 1.

I added an example GDS file in the EBeam PDK Examples/ebeam folder.

closed 2018-08-16 by lukasc-ubc