SiEPIC/SiEPIC_EBeam_PDK #188: Verification Error in Klayout

MohamedElRayany, 2018-05-13:
Hi all,

After finishing layout and doing verification to check for errors, I got connectivity errors. I am using script provided by SiPIC 0.3.43 library for snapping components and verification. Can anyone help me in knowing why I received this error?
To understand it more clearly, please see the following pictures of the connection
https://ibb.co/fLE5Dy
https://ibb.co/b6ObLd
https://ibb.co/fO77Ad

MohamedElRayany, 2018-05-22:
On a side note, I hope it will be useful, this problem appeared to me when I was using the last version of Klayout (0.25.3) on Windows 10. I didn't face similar error when I created the same structure using older versions of Klayout.

lukasc-ubc, 2018-08-16:
Hi @MohamedElRayany 

Does the problem still exist? I believe we may have fixed it.  If it is still there, can you please post the GDS file that has this problem?

thank you
Lukas

MohamedElRayany, 2018-09-23:
Hi @lukasc-ubc 

Thanks for your response and, yes, the problem still exists.

I am using [the script provided by Lumerical](https://www.lumerical.com/support/video/gds-export.html)'s team that is used to export the GDS-II layout from one of their tools (FDTD or MODE, for instance). After that, I used Klayout to connect the exported layout with the grating couplers and other structures that can be found in SiEPIC-EBeam library, but it gives me an error message as shown below:
[image]
[image]
[image]
[image]

Hope to find a solution to this problem soon.

lukasc-ubc, 2018-09-27:
Hi @MohamedElRayany 

The verification is there to detect if mature components are not connected.  What you are doing is creating your own components from polygons.  In that case, you need to encapsulate the GDS polygons from FDTD or MODE into a component (add pins, etc).

This is described in the section for Developers: https://github.com/lukasc-ubc/SiEPIC-Tools/wiki

specifically: https://github.com/lukasc-ubc/SiEPIC-Tools/wiki/Component-and-PCell-Layout

On a more general point, I'm not sure why you would create waveguides in FDTD/MODE and import them into KLayout...  I would just draw them natively in KLayout.

closed 2023-11-21 by lukasc-ubc