SiEPIC/SiEPIC_EBeam_PDK #190: Bragg Grating Issues

amistry-ubc, 2018-05-29:
Hey guys,

I noticed 2 errors with the bragg gratings in the IME and Ebeam PDKs.

1) Odd numbers corrugation widths (51 nm, 53 nm, etc.) round to the lowest even numbers width (50 nm, 52 nm, etc).

2) For all bragg gratings we have with period = 317 nm, one half period section is 158 nm wide and the other is 159 nm wide, which leads to a non-uniform grating. This is the case for all odd period numbers.  I suppose this is due to the 1 nm .gds grid spacing.

Thanks,

Ajay

StephenHLin, 2018-05-30:
is this the straight bragg grating you are talking about?
I found this on some of my devices where I had coordinates that were smaller than the DBU unit, they would round up or down occasionally.

Is your klayout's grid measurements set to 0.001um ? That could help narrow down the reason for this.

mustafacc, 2018-05-30:
The first error is fixed (https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/commit/f8a94661336c9758ec10378ef32b17084bbc7a11)

About the second error with non equal half period sections, its not possible to have two equal sections of 158.5nm since the minimum grid is 1 nm, using even periods will ensure equal half-period sections

closed 2018-08-16 by lukasc-ubc