SiEPIC/SiEPIC_EBeam_PDK #191: Klayout 0.25 integration with Lumerical Interconnect

liaquat85, 2018-08-10:
Dear Lukas, 

I tried to connect the Klayout 0.25 with Lumerical Interconnect. I followed the procedure as given in 
https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK/wiki/Installation-instructions

When I try to connect the KLayout with Lumerical Interconnect I am given this error 

[image]

What should I do ?

lukasc-ubc, 2018-08-12:
Hi @liaquat85,
Can you please let me know which versions of all the softwares you are using?  (make sure they are the latest ones).  The error you are seeing is in INTERCONNECT's python integration script (lumapi.py, line 87).  
I'll try in the next day or two to see if I get this error in the latest versions...

lukasc-ubc, 2018-08-12:
Can you also please send me the GDS file that you are trying to simulate? I would like to see if I can reproduce your problem.

liaquat85, 2018-08-13:
Hi  Lukas,

 

This is the file I want to simulate. 

 

Regards, 

Liaquat Ali

liaquat85, 2018-08-13:
Hi Lukas ,

 

I am using Klayout 0.25.3

Lumerical Interconnect  7.0.1332

Regards, 

Liaquat Ali

lukasc-ubc, 2018-08-13:
Can you please try INTERCONNECT 8?

liaquat85, 2018-08-16:
Hi  Lukas,

 

Can you please tell how to referecen the Ebeam library PDK if one is using it in his research work ? 

 

Regards, 

Liaquat Ali

lukasc-ubc, 2018-08-16:
Hi @liaquat85,

The version of the Lumerical tool you are using is quite old, so I suspect that the issue is that the Python integration is outdated.  Try the latest version.

For your question about referencing the tool, do you mean what to put in a publication?  There is a paper suggested for citing this work on the main page: https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK
You can also put the GitHub EBeam URL in the references section.

regards
Lukas

gitelfin, 2018-08-20:
Does this application requires specific license ?? I got the following error:
[image]

lukasc-ubc, 2018-08-20:
Yes there is a KLayout integration license feature

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

closed 2021-07-09 by lukasc-ubc