SiEPIC/SiEPIC_EBeam_PDK #202: different Layout of directional coupler on a Mac El Capitan

yaohuic, 2019-03-02:
1. I was testing Klayout with SIEPIC 0.3.52 on my old El Capitan Mac Pro. I found that the following default Ebeam Pcell "ebeam_dc_te1550" has a different shape as below. Lc=10um.  The "S-bend" on each port now looks two 90-degree "ARC", which takes a large area as 30.1um x 22.2um. I am not sure what is the root cause, or a particular software-compatible/python-compatible issue with an old Mac OSX.

2. The layout on my windows laptop and (the fabricated layouts passed through edx layout check) shows 22.1um x6.2um dimensions with slimmed "S-bend". Such Pcell different between my Mac and windows machine may lead me to inconsistent layout works. 

Hence, I am very interested to understand why it happens.  Any minimum Mac python packages version need to be confirmed? 

[image]

lukasc-ubc, 2019-03-03:
Can you send me the screenshot for the one that looks different?

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

yaohuic, 2019-03-03:
Follow is the screenshot on El Captian Mac:
default  ebeam_y_1550 vs. "Ebeam_dc_te1550"
[image]

yaohuic, 2019-03-03:
Following is screenshot on windows laptop: (The standard layout look)
default ebeam_y_1550 vs. Ebeam_dc_te1550, both have similar vertical distance between two ports. 
[image]

yaohuic, 2019-03-04:
And I have an minor question about layer colors and filling patterns through Klayout:

1.  Feels like Mac has better visual effects on the filling patterns of SiEPIC PDK;
2.  Klayout on Windows normally does not show the relevant filling patterns;
3.  Is it a Klayout (windows vs. Mac) issue?  or "color/graphic coding difference between windows and Mac?

lukasc-ubc, 2023-11-21:
Please try turning off "Show Layers without Fill"
[image]

closed 2023-11-21 by lukasc-ubc