SiEPIC/SiEPIC_EBeam_PDK #205: Cannot add extensions

SRToyota, 2019-09-10:
I'm trying to add the packages on KLayout running the Ubuntu 18 version, and I'm receiving the following error:

Error fetching spec file for package 'siepic_tools': Error 99: Network API error, fetching https://github.com/lukasc-ubc/SiEPIC-Tools.git/tags/v0.3.56/klayout_dot_config/grain.xml

Thanks for your help,

lukasc-ubc, 2019-09-10:
Hi Sean,

That usually comes if the internet connection is not working. Can you check that first?  Can you also check on a different operating system to see if the problem persists?

SRToyota, 2019-09-10:
Hi Lukas,
Thanks for your quick response. That’s what I thought the error meant, but I tried installing it on a windows machine, on the same network and it seems to be fine.
I’m traveling the next 3 weeks and I was trying to limit how many laptops I’m carrying with me. It doesn’t sound like it will be an easy fix, so no worries, I’ll just try something else out.

Thanks,

SRToyota, 2019-09-10:
> Hi Sean,
> Sounds like this is a problem with KLayout, rather than SiEPIC-tools. In that case, can you please check other packages, and if the problem exists for all, then I suggest posting it on the KLayout GitHub forum. https://github.com/KLayout/klayout/issues <https://github.com/KLayout/klayout/issues>

Hi Lukas,
You were right, I’ll post to the Klayout forums. I appreciate your help