SiEPIC/SiEPIC_EBeam_PDK #225: ebeam_wg_strip_1550 CML problem

lukasc-ubc, 2020-05-25:
Component needs to be deleted or fixed in the CML.

[image]

galaturka, 2020-05-25:
Hello Mr. Chrostowski,

How can we fix it in CML ? Should we replace old one with an updated version, or ?

Thanks for your support.

lukasc-ubc, 2020-05-25:
I have used that component for a long time.  You are the first to ask...  

Please use the "ebeam_wg_integral_1550" component. It is also the one that is used when we make waveguides in KLayout.

But as for fixing, yes, copying the integral version to this one would fix it.  or we can just delete this old one.