SiEPIC/SiEPIC_EBeam_PDK #229: Obsolete FDTD path, Ebeam_dc_halfring

atofini, 2020-10-05:
Lumerical 2020a path has been changed.  The script path for compact models needs to be changed in order to automatically generate fdtd simulations for new S-parameters.

Example:

 Old Path `Lumerical/fdtd/bin/fdtd_solutions.exe`
New Path `Lumerical/2020a/bin/fdtd_solutions.exe`

lukasc-ubc, 2021-06-15:
Solution from @jaspreetj 

[image]

The problem is caused by the FDTD directory location. Here is how you can quickly fix it:
In interconnect, right click on custom and redirect to the directory where you unpacked the cml library.
 
Drag and drop the component you would like to edit from the library in custom section.
Right-click the component and click "edit"
In the Scripts>setup section, scroll to line #125 for windows (or 128 or 131) based on the OS, and replace the FDTD path to correct one.
For windows, the file is located at " C:\Program Files\Lumerical\FDTD\bin\fdtd-solutions.exe "
This fixed the issue for me

lukasc-ubc, 2021-06-15:
It looks like for some people Lumerical installs in a folder named v211.  So you need to check where FDTD is located and update your script accordingly.

closed 2021-06-17 by jaspreetj