SiEPIC/SiEPIC_EBeam_PDK #233: How to modify the SiEPIC_EBeam_PDK

Zwlzzm, 2020-12-24:
Dear Professor Lukas,

The SiePIC_EBeam_PDK is very useful, especially the automatic waveguide connection function. I want to modify your pdk to make it suitable for us. I currently feel that there should be two pieces. 
The first is pcell. How to build a new one on the basis of your pdk 
And how to add a new compact model? Which folders should I look for? Can you give me some suggestions or related web pages?
Thank you very much!

Looking forward to hearing from you!
Best regards!

jaspreetj, 2021-01-08:
Hi @Zwlzzm ,

All layouts (fixed and pcells) reside inside the `klayout_dot_config>tech>EBeam>pymacros` folder. If you are looking to build new pcells, I suggest having a look at examples in the `pcells_EBeam` folder. 

Let me know if you have any other questions.

Zwlzzm, 2021-01-18:
@jaspreetj ，

Thank you very much, I have found it！
Best regards!

lukasc-ubc, 2021-01-18:
You can also take a look at the Wiki page for SiEPIC-Tools

Zwlzzm, 2021-01-18:
@lukasc-ubc,
Dear Professor Lukas,

Thank you for your advice.
I will have a  look at it and study it.
Thank you！

closed 2021-01-21 by Zwlzzm