SiEPIC/SiEPIC_EBeam_PDK #234: How to run install SIEPIC PDK in Klayout and how run circuit simulation

Zwlzzm, 2021-01-20:
@lukasc-ubc ,
Dear Professor lukas,

I have to install KLayout and SIepic PDK in Centos 7. And I found something strange.
Could you give me some assitance?

1.  I tried to install Siepic PDK on centos 7, but it always shows there is  no python 3, although I have installed python3.

[image]

[image]

(I install Siepic PDK by copying the files of  "klayout_dot_config" in siepic-tools and sepic_ebeam_pdk to the new folder "salt" in .klayout (because I cannot install it from Tool->Manage Package, it can’t be installed successfully);)
[image]

2. Beside that, I found it come out a lot of errors when I open KLayout with terminal. What cause this?
[image]

3. We purchased the lumerical education version and the lumerical-KLayout Interoperability license.
Those licenses are actived.
Then how to use it for circuit simulation?

a。I click simulation, Circuits->Set up lumerical Interconnect Compact Model, it always prompts me to find the installation location of lumerical. But  I don't know where is it.  Do you know where to find this link?
[image]

b，I tried to pull two structures from the Siepic PDK (all the structures have port names), and click simulation, Circuits-> Circuit simulation Lumerical Interconnect, but it always prompts me that I should have an opt_in label, so what is wrong? 
[image]

[image]

Could you give me some suggestions?

Looking forward to hearing from you!
Best regards!
Weiling

Zwlzzm, 2021-01-21:
The interconnect versions are 9.6.2466, 
[image]
The lumerical-KLayout Interoperability license seems to be this:
[image]

The Klayout version are 0.26.9

Zwlzzm, 2021-01-21:
> The interconnect versions are 9.6.2466,
> 
[image]

> The lumerical-KLayout Interoperability license seems to be this:
> [image]
> 
> The Klayout version are 0.26.9

Zwlzzm, 2021-01-21:
> @lukasc-ubc ,
> Dear Professor lukas,
> 
> I have to install KLayout and SIepic PDK in Centos 7. And I found something strange.
> Could you give me some assitance?
> 
> 1. I tried to install Siepic PDK on centos 7, but it always shows there is  no python 3, although I have installed python3.
> 
> [image]
> 
> [image]
> 
> (I install Siepic PDK by copying the files of "klayout_dot_config" in siepic-tools and sepic_ebeam_pdk to the new folder "salt" in .klayout (because I cannot install it from Tool->Manage Package, it can’t be installed successfully);)
> [image]
> 
> 1. Beside that, I found it come out a lot of errors when I open KLayout with terminal. What cause this?
>    [image]
> 2. We purchased the lumerical education version and the lumerical-KLayout Interoperability license.
>    Those licenses are actived.
>    Then how to use it for circuit simulation?
> 
> a。I click simulation, Circuits->Set up lumerical Interconnect Compact Model, it always prompts me to find the installation location of lumerical. But I don't know where is it. Do you know where to find this link?
> [image]
> 
> b，I tried to pull two structures from the Siepic PDK (all the structures have port names), and click simulation, Circuits-> Circuit simulation Lumerical Interconnect, but it always prompts me that I should have an opt_in label, so what is wrong?
> [image]
> 
> [image]
> 
> Could you give me some suggestions?
> 
> Looking forward to hearing from you!
> Best regards!
> Weiling

lukasc-ubc, 2021-01-21:
Did you solve it? I see that you closed it?

Zwlzzm, 2021-01-21:
@lukasc-ubc , I click something wrong, so it shows closed. it is not solved

Zwlzzm, 2021-01-25:
@lukasc-ubc ,
Dear Professor Lukas.
Could you give me some suggestions?

jaspreetj, 2021-01-26:
Hi @Zwlzzm, 

1. Which python version is detected by Klayout? You can see this by going to  `Help>About` section in Klayout.
2. Could you try installing siepic-tools with python version 3.6.9 or 3.7.3(for Klayout 0.26.9)? 
3. As for finding lumerical installations - this should be in `/opt/lumerical/<version>`. 
4. In order to run simulations, you must use either detector and laser components from `Instance>SiEPIC General - Laser, Detector` library, or a text label which has a prefix of opt_in_your_custom_label_identifier.

Let me know if you encounter any issues.

Zwlzzm, 2021-01-27:
Hi, @jaspreetj ，

Thank you for your kind respond.

1,2  The version detected by KLayout is 0.27.5. As you suggested, I have modified the default python format of centos to python 3.7.3.
Do you know how to match this two?
[image]
[image]

3. Regardless of these errors carried by python, I tried to locate the Lumerical in  /opt/lumerical/v202, however, it still come out error. What could be this caused by?
[image]
[image]

4.  According to what you suggested, I pull a laser from Instance>SiEPIC General - Laser,  I don't know if it shows normally.  Then, I click  simulation, Circuits-> Circuit simulation Lumerical Interconnect, it shows there is no component. 
[image]

Beside, why cannot  I do circuit simulations using  two grating couplers  and a waveguide from Ebeamv3.30？
In other word, there are only 2 components (detector and laser) can be used to do cicuit simulation?

Looking forward to hearing from you!
Best regards!

Zwlzzm, 2021-01-28:
@lukasc-ubc ，
@jaspreetj ，

Supplementary explanation, I alao tried to install SIepic Tools from KLayout, however,  it always shows that this page is not found.
[image]
https://github.com/siepic/SiEPIC-Tools.git/!svn/bc/2345/tags/v0.3.71/klayout_dot_config/pymacros/Keybindings/Simulation%2C%20Components/

It seems that this  link is invalid. I also cannot install it by this method.

lukasc-ubc, 2021-03-09:
- to get support for these tools you can join courses, at www.siepic.ca/education, or the company SiEPIC Kits Ltd can provide commercial support: siepic.com
- I suggest starting with an example layout that is know to work, such as the SiEPIC > Example Layouts > Test structures: Double-bus ring
[image]
- then export the netlist (rather than trying to run INTERCONNECT directly) 
[image]
- then load the .spi file in Lumerical INTERCONNECT
- make sure you have the EBeam CML library installed in INTERCONNECT

Zwlzzm, 2021-03-09:
Dear professor Lukas Chrostowski, 

Thank you very much for your reply and your kindness;

Thank you for your reminder. I know these tools after attending your course on edx, but I didn’t know much about the combination of sipeic and interconnect at that time;

According to your method, in the third step, use the Lumerical interconnect software to open the .lsf file generated in the second step and run it. It will automatically import the .spi file and generate simulation results;

Follow the method, on windows, everything works well;
On centos, a small error were reported during the process (circuit simulation-&gt;export files only), and then by adding "int" to scripts.py, it can generate files normally and the simulation results will appear;

According to your understanding, if I can run successfully according to the method you gave, does it mean that the license(generic_netlist _import) is already in use?

What could be the reason why I can't directly click the interconnect in klayout to perform simulation? Or is this method you given above the correct way to use this function (circuit simulation)?

Thanks again for your reply.

Best regards!

  

 
I suggest starting with an example layout that is know to work, such as the SiEPIC &gt; Example Layouts &gt; Test structures: Double-bus ring

lukasc-ubc, 2023-11-21:
@Zwlzzm 
Can you please check if the installation problem has been resolved in the more recent versions of SiEPIC-Tools?
thank you
Lukas