SiEPIC/SiEPIC_EBeam_PDK #235: Why my EBeam PDK libraries lost even if I install EBeam_PDK 0.3.32

zhangxlei, 2021-04-22:
@lukasc-ubc 
Dear Professor lukas,

I have installed klayout(0.26.11) and Siepic_EBeam_PDK(0.3.32) in Windows 10. I have used the software and your PDK well. 

But today after I update the Siepic_EBeam_PDK, the SiEPIC_EBeam_PDK libraries is lost. 
I have change the version of klayout from 0.26.7 to 0.26.11 and reinstalled the package the libraries still lost.
[image]
[image]

In addition, the SiEPIC-Tools has no problem.
[image]

Could you give me some suggestions?

Looking forward to hearing from you!
Best regards!
Weiling

lukasc-ubc, 2021-04-23:
Could you please check the error log?  Under the File menu.  Perhaps there are errors when initializing the libraries.

zhangxlei, 2021-04-23:
Thanks for your suggestion!
There is a warning or error in log viewer.
[image]
But how to solve the problem?

zhangxlei, 2021-04-23:
In addition, I want to install klayout and the PDK on another computer, it had the same problem and error in log viewer.

lukasc-ubc, 2021-04-23:
That’s an easy fix. 

In that file, add a line at the top:

from pya import Path

You’ll need an external editor since the file is read only in KLayout when in the salt folder. 

Lukas Chrostowski
Professor, Electrical and Computer Engineering
University of British Columbia
Quantum Matter Institute
Member, College of the Royal Society of Canada
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca
https://qmi.ubc.ca/about/directory/lukas-chrostowski

lukasc-ubc, 2021-04-23:
actually, we need Path, Point, etc.

So perhaps:

from pya import *

probably this needs to be fixed in other files too.

lukasc-ubc, 2021-04-23:
I released v0.3.33. can you see if that fixes it? thanks.

zhangxlei, 2021-04-23:
I have tried add a line at the top: from pya import Path/ from pya import *, it seems to make no difference.

zhangxlei, 2021-04-23:
I have updated your new release, the problem still exists.

zhangxlei, 2021-04-23:
it seems not to add a line at the top, I add the code into the second line. It is right?
[image]

zhangxlei, 2021-04-23:
After I add the code into second line, the log viewer has no error or warning but the library still lost.

zhangxlei, 2021-04-23:
I have changed the version to 0.3.30, the libraries reappears and the problem is solved. I guess there is something the matter with your code of 0.3.32.

ANT-Jocelyn, 2021-06-07:
@lukasc-ubc I'm having this same issue with the current version 0.3.34. The library import scripts are no longer running automatically successfully when KLayout is opened. They can be run manually from within KLayout each time the software is opened to show the libraries but you have to accept the following exception: 
[image]

mustafacc, 2021-06-07:
Hi @ANT-Jocelyn  , which version of KLayout are you currently using?

ANT-Jocelyn, 2021-06-07:
@mustafacc 0.26.12

lukasc-ubc, 2021-07-09:
Hi @ANT-Jocelyn 
sorry for the slow response.  Do you still have this problem, say in SiEPIC-Tools 0.3.79 and the SiEPIC-EBeam-PDK 0.3.36?  If it works with the EBeam PDK, compare your library loading code with the EBeam one.

ANT-Jocelyn, 2021-07-20:
@lukasc-ubc @mustafacc this error seems to be fixed in the SiEPIC Tools 0.3.79 and EBeam-PDK 0.3.36 releases, thanks for checking.

closed 2021-07-21 by mustafacc