SiEPIC/SiEPIC_EBeam_PDK #240: Why the transmission auto generated by the code is different from selecting port and "visualize->S/T"

Zwlzzm, 2021-07-08:
Hello,

I was confused by this by a long time.

In function "Component: FDTD simulation for S-Parameters, create Compact Model"of siepic ebeam pdk, there is a code to plot transmission in dB.

lumapi.evalScript(_globals.FDTD, " \
              plot (wavelengths, 10*log10(abs(Sparam(:,%s))^2),  'Wavelength (um)', 'Transmission (dB)', 'S_%s_%s, mode %s'); \
               " % (modes.index(m)+1, p.pin_name, in_pin.pin_name, modes.index(m)+1) )

From this script, it seems that it plots transmission of {S31}2  and  {S41}2 using "10*log10(abs(Sparam(:,%s))^2".

For a dc design, the transmission auto generated is shown like this:
[image]

However, when I plot {S31}2 ,  {S41}2, T31 and T41 by selecting port, and choosing "visualize->S/T", for the same design, I got transmission differ form the figure auto generated.

[image]
Even if they are all converted to the dB scale, their values are not the same. Why?

So what does the ordinate of the curve generated by scipt represent?

Anyone who know this?

lukasc-ubc, 2021-07-09:
I suggest plotting the bottom graphs in the dB scale.  What your two graphs show are the linear values, and furthermore the Real part of the field (Scalar operation).  You need to plot Abs^2 of the S-parameters by changing the Scalar operation. Then convert to dB (which isn't possible in the Lumerical GUI).

Zwlzzm, 2021-07-09:
@lukasc-ubc,

Thank you very much for your respond.
I am sorry, but I don't really understand, the bottom graphs are already showing Abs^2 of S parameter.

I plot the left bottom graphs in the dB scale,
for port3,
[image]
for port4,
[image]

So it seems that the transmission of auto generated are different from that of  Abs^2 of S parameter, even they are plotting the same thing and they are from the same FDTD file.

I am very confused with it. What is the transmission of auto generated stand for? Why they are different?