SiEPIC/SiEPIC_EBeam_PDK #242: Why does the mesh not cover the entire polygon？

Zwlzzm, 2021-07-09:
Hi,
In function "Component: FDTD simulation for S-Parameters, create Compact Model" of sipeic ebeam pdk,
why does the mesh not cover the entire polygon, but only a small part? Are there any considerations for this?

    lumapi.evalScript(_globals.FDTD, " \
      newproject; closeall; \
      addfdtd; set('x min',%s); set('x max',%s); set('y min',%s); set('y max',%s); set('z span',%s);\
      set('force symmetric z mesh', 1); set('mesh accuracy',1); \
      set('x min bc','Metal'); set('x max bc','Metal'); \
      set('y min bc','Metal'); set('y max bc','Metal'); \
      set('z min bc','%s'); set('z max bc','%s'); \
      setglobalsource('wavelength start',%s); setglobalsource('wavelength stop', %s); \
      setglobalmonitor('frequency points',%s); set('simulation time', %s/c+1500e-15); \
      addmesh; set('override x mesh',0); set('override y mesh',0); set('override z mesh',1); set('z span', 0); set('dz', %s); set('z', %s); \
      ?'FDTD solver with mesh override added'; " % ( FDTDxmin,FDTDxmax,FDTDymin,FDTDymax,FDTDzspan, \
         Z_symmetry, FDTD_settings['Initial_Z-Boundary-Conditions'], \
         wavelength_start,wavelength_stop, \
         FDTD_settings['frequency_points_monitor'], sim_time, \
         FDTD_settings['thickness_Si']/4, FDTD_settings['thickness_Si']/2) )

I see there are some sentence, are this becauses of convergence testing?

definitions:
  z: thickness direction
  z=0: centre of the waveguides, so we can use symmetric mesh in z
       which will be better for convergence testing as the mesh won't move

lukasc-ubc, 2021-07-09:
The simulation mesh covers the entire volume ("addfdtd"). 

It is the mesh override that you are concerned with ("addmesh"). This one is configured to have higher resolution in the thickness direction only.

Being careful with the mesh is important when doing convergence testing, otherwise a changing mesh will give you slight different results.

Zwlzzm, 2021-07-09:
@lukasc-ubc ，

Thank you very much for your respond.
You mean, the mesh override is only assigned to z direction. But  I see there are still value in x span and y span of mesh overrided. Does those  value don't  take effect and are assigned randomly?

In addition, I would like to ask about this mesh, z=si_thickness/2,dz=si_thickness/4, is it good enough for most of convergence test?