SiEPIC/SiEPIC_EBeam_PDK #250: Installation instructions - Part 3 Lumerical Interconnect

langrock, 2022-09-19:
Following part 3 of the installation instructions, I get to the point where INTERCONNECT opens, but this is followed by the termination of both INTERCONNECT and Klayout. I am using the current version of Lumerical, and version 0.27.5 of Klayout; I also tried 0.27.11 with the same result. Is there a log file one could look at to see why this isn't working? The instructions make it seem as if some sort of confirmation should be displayed and that one would need to manually close INTERCONNECT. I would certainly not expect Klayout to self terminate. Thanks

lukasc-ubc, 2023-11-08:
Sorry for the late response.  Would you be able to try KLayout 0.28.x?

lukasc-ubc, 2023-11-08:
I found an issue with this, which I fixed in SiEPIC-Tools 0.4.5 and SiEPIC-EBeam-PDK 0.3.57.

Thank you for reporting this.

closed 2023-11-21 by lukasc-ubc